SpacehuhnTech/esp8266_deauther #1434: ESP8266 does not detect clicks from buttons.

suchencjusz, 2021-05-20:
Hello, I have a problem with esp8266 buttons.
Buttons are connected to GND and digital ports (D5-D7),
esp don't detect the clicking, also I connected buttons to VCC, it did not work.
(SSD1306 I2C OLED works fine)

I'm using nodemcu 0.9

[image]
(maybe useful photo)

Torepda, 2021-05-20:
For me, uploading the code again helped me. It is best to set it when uploading ,,earse all data''

suchencjusz, 2021-08-18:
co kurwa stale

closed 2021-11-24 by stale[bot]

OCEANOFANYTHINGOFFICIAL, 2022-02-20:
spacehuhntech please help me, i am facing the same problame

Torpeda0, 2022-02-23:
Have you connected the buttons to GND?

OCEANOFANYTHINGOFFICIAL, 2022-02-24:
> Have you connected the buttons to GND?

Obviously, i have done the same procedure mentioned in the diagram. But the buttons are not working, my nodemcu and display is working fine. But the buttons are not working. If you know why then please help me. It is my office project.