SpacehuhnTech/esp8266_deauther #1444: Unused GPIO pins ?

Volus208, 2021-06-12:
Hello,
I have a deauther watch v1 (the one with the breakout gpios) that I really like, especially since it is open source.
I would like to add a vibrator that I got from an old smartphone, to which I added a mosfet so it doesn't draw too much power from the GPIO, but I can't find any unused gpin to control it.
Would the highlight LED pin work ? Would I need to desolder the LED first ?
Thanks in advance

Zorganight, 2021-06-30:
Hello,
I recommend **not** to desolder LED, which card are you using?

Volus208, 2021-07-01:
Well, I already desoldered it and soldered the vibrator to it, and it works 😅
If I understand correctly, my card is a Dstike deauther watch v1

Zorganight, 2021-07-02:
if its work no problem 😄

closed 2021-07-02 by Volus208