SpacehuhnTech/esp8266_deauther #1452: NodeMCU oled example resets every few seconds, checked code like described in the wiki

strolll395, 2021-07-05:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

**Describe the error**
A clear and concise description of what the problem is.
What do you think causes it?

```
Error/Compile/Output Log
Mounting SPIFFS...OK
Formatting SPIFFS...OK
Formatting EEPROM...OK
Loading settings...Invalid Hash
Settings reset to default
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: true, captive-portal: false
STARTED! \o/
2.6.0

 ets Jan  8 2013,rst cause:4, boot mode:(3,4)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

Mounting SPIFFS...OK
Loading settings...OK
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: true, captive-portal: false
STARTED! \o/
2.6.0

**To Reproduce**
Steps to reproduce the behavior:
1. Connect OLED
2. Upload OLED example
3. Open serial monitor on baud 115200
4. Device resets every few seconds without any output on the OLED.

**Expected behavior**
A clear and concise description of what you expected to happen.
OLED should display output and be controllable.
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: [Windows 10]
 - Version [Deauther 2.6]
 - Hardware [AZDelivery NodeMCU]

**Attempts**
What have you already tried and didn't work out.
Changed code, didn't help.
**Additional context**
Provide as much information as possible, better too much than too little!
If you don't use this template, your issue might be closed and tagged invalid!

strolll395, 2021-07-05:
Oh, and I2C Display example is selected as the board, checked that.

closed 2021-07-10 by strolll395

captclearleft, 2021-12-09:
Was this issue closed with a solution - I dont see the solution.  I will continue to look.  

I am having the same exact issue.  Following the "**Setup Display & Buttons**" exactly. (NodeMCU with Display Example I2c) 
I am using the LOLIN NodeMCU v3 with the ch340 chip (I have 3 of them, and have tried 2 of em).
I tried changing the code for the SDA and SCK pins from 5, 4 to 4,5 to D1,D2.  I tried swapping those wires.  
I tried uncommenting these lines in the A_config.h... 
       #define NODEMCU
       #define DISPLAY_EXAMPLE_I2C
I tried different power sources.
I tried just flashing it with the Node MCU Flasher and the .bin file.

I don't have any custom anything going on, just set up exactly as described in the tutorial.  

The regular NodeMCU code without display works flawlessly on both boards.  
When I run the OLED display code from https://github.com/squix78/esp8266-oled-ssd1306 The display works fine.  No resets, and the display shows everything just fine.   

I am guessing its something to do with the way this board and the ch340 chip exists.  
Maybe ill try the resistors on D4 and D8.

Any one else use this board, or have a solution?  Thanks so much.

captclearleft, 2021-12-09:
Just another update.  Tried adding the resistors (D4 - High, D8 - Low, and I even tried Pulling Reset High (10k))
Same results - No display, constant resets... 

wdt reset
load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v000cb3d0
~ld

Reloaded the SSD1306SimpleDemo.ino sketch.  Screen still works fine.  
I still think its something screwy with this ch340 chipped LOLIN NodeMCU.  

I did try my Adafruit feather esp8266 - That didnt work either.  I may have not hooked that up correctly as there are only the two SDA and SCK (4,5) on that board and thats what I used... 

I updated all my libraries for Arduino, still no luck.  Any ideas would be appreciated.  Thanks

strolll395, 2021-12-11:
@captclearleft 
i used this code back in july, display and buttons work just fine. Remember to adjust your settings accordingly 
https://github.com/wh1ter0z?tab=repositories
(esp8266 deauther)

captclearleft, 2021-12-11:
@strolll395 Thanks.  You didn't mention what was the fix for you?  
Was it a setting - Which setting?  I have used the defaults and the only change I need to make is the screen.  I am using 
#define SSD1306_I2C  Everything else is as you have on your fork.

Which NodeMCU are you using.  I think my issue is being caused by the fact that  I am using the NodeMCU by Lolin that has the cheap CH240 USB chip...  

Thanks

captclearleft, 2021-12-11:
Ok, So I copied your A_config.h into the Arduino IDE replacing the original A_config.h code.
Used "NodeMCU" As the "Board" and the "Deauther Config" under tools.  
And it worked.  
It looks like there is something in the #if #elseif statements when using the "Deauther Config= Deisplay Example IIC" mode that causes an issue.  Using just the file with all the #if #elseif statements removed and then just defining the display worked.
Thanks
https://github.com/wh1ter0z/esp8266_deauther

ASCII-encrypt, 2023-02-26:
so ive done everything here and no more bootlooping but i still dont have any screen tried changing values still nothing

1jammer1, 2023-02-27:
have you tried changing the hardware

-- 

_____________________________________________________________

CONFIDENTIALITY/PRIVACY NOTICE: This message and any attachments 
transmitted
with it is for the designated recipient only and may contain 
privileged or
confidential information. If you have received it in error 
please notify the
sender, via return e-mail, immediately and permanently 
delete the original. Any
unauthorized review, disclosure, dissemination, 
distribution, or copying of this
e-mail is strictly prohibited. Thank you.

ASCII-encrypt, 2023-02-28:
> have you tried changing the hardware
> 
> 
> > so ive done everything here and no more bootlooping but i still dont have
> > any screen tried changing values still nothing
> >
> >
> 
> 
> 
> 
> CONFIDENTIALITY/PRIVACY NOTICE: This message and any attachments 
> transmitted
> with it is for the designated recipient only and may contain 
> privileged or
> confidential information. If you have received it in error 
> please notify the
> sender, via return e-mail, immediately and permanently 
> delete the original. Any
> unauthorized review, disclosure, dissemination, 
> distribution, or copying of this
> e-mail is strictly prohibited. Thank you. 

Didn't have to breadboard started to smoke got a new one and it's all working now 

Safe to say that was the issue

elimerins, 2023-07-10:
> @captclearleft i used this code back in july, display and buttons work just fine. Remember to adjust your settings accordingly https://github.com/wh1ter0z?tab=repositories (esp8266 deauther)

my bug fixed.