SpacehuhnTech/esp8266_deauther #1453: How to reduce the brightness of battery level indicator green LED?

RahulNambi, 2021-07-16:
Hello Guys,
Which defined MACRO should i check in code to modify to reduce the brightness of green led glowing ?

Thanks,
RNambi.

tubblindo, 2023-05-29:
Did you manage to solve it? I have the same problem but I don't know how to do it

lspoplove, 2023-05-29:
Did you manage to solve it? I have the same problem but I don't know how to do it

modem-man-gmx, 2023-11-02:
same her, but ... there is no way to reduce it.
the 4 LEDs are connected to injoinic IP5306
[Datasheet](http://www.injoinic.com/wwwroot/uploads/files/20200221/0405f23c247a34d3990ae100c8b20a27.pdf)
This circuit handles each 2 of the 4 LEDs anti-parallel, to control the LEDs with only 3 wires. There is no common resistor or such, if wired with 4 LEDs.

I don't understand the description about the pwr switch completely. 

Push button’s connecting is shown in Fig 5. IP5306 can identify long push and short push.

- If button is pushed longer than 30ms but shorter than 2s, IP5306 will identify the action as short 
push. Short push will open SOC indicator LEDs and step-up converter
- If button is pushed longer than 2s, IP5306 will identify the action as long push. Long push will
close step-up convertor, SOC indicator LED and flashlight LED.
- If button is pushed shorter than 30ms, IP5306 will ignore the action.
- If two short push is detected within 1s, IP5306 will open or close flashlight LED

Almost all actions I tried just cut or reconnect the power from ESP and OLED.

My best LED dimmer so far:
small piece of thin paper and a stripe of transparent TESA duct tape...