SpacehuhnTech/esp8266_deauther #1464: evil twins phishing can upload html page same as lapara master wifi

anonymousseafarer, 2021-08-12:
**Do similar feature requests issues already exist (open and closed)?**
If yes, please link them here:

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.
If you don't use this template, your issue might be closed and tagged invalid!

ngocthuong234, 2021-08-17:
With only one esp8266, deauth + wifi password + Facebook, Google, Twitter,..
https://youtu.be/oKsRb4KXXk0
https://youtu.be/Nt-0aTfPECQ

anonymousseafarer, 2021-08-17:
Can you send me the bin file for this firmware? Please

cakralithium, 2022-02-14:
> Hanya dengan satu esp8266, deauth + kata sandi wifi + Facebook, Google, Twitter,.. [https://youtu.be/oKsRb4KXXk0 ](https://youtu.be/oKsRb4KXXk0) https://y

> Hanya dengan satu esp8266, deauth + kata sandi wifi + Facebook, Google, Twitter,.. [https://youtu.be/oKsRb4KXXk0 ](https://youtu.be/oKsRb4KXXk0) https://youtu.be/Nt-0aTfPECQ

can i give your firmware in zip folder..