SpacehuhnTech/esp8266_deauther #1486: ESP8266 ESP 12S USB - flashing and webpage looking fine - but not de-authering of an AP

polleke67, 2021-10-14:
> Have you searched for existing (open and closed) issues describing the same problem?
Yes

** A clear and concise description of what the problem is **
1. I purchased an ESP8266 ESP 12S USB 4 MB NodeMCU at https://nl.aliexpress.com/item/32822640337.html?gatewayAdapt=glo2nld&spm=a2g0s.9042311.0.0.15f54c4dFqrkZX
2. Download and installed USB UART drivers "CP210x_Universal_Windows_Driver.zip" (v10.1.10 1/13/2021): https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers
3. Download and installed "NODEMCU ESP8266Flasher.exe" https://github.com/nodemcu/nodemcu-flasher/blob/master/Win64/Release/ESP8266Flasher.exe
4. I used "esp8266_deauther_2.6.1_NODEMCU.bin" at offset 0x00000 to flash it.
5. I was able to access 192.168.4.1 and could select an AP to de-auther it, but I was still able to connect by mobile to that AP. The LED on the USB dongle was not blinking or switched on.

** What do you think the problem is? **
Hardware not compatible with the selected firmware or release. Maybe use another firmware or release?

**Environment (please complete the following information):**
WIFI Deauth Detector met Case ESP8266 ESP-12S USB NodeMCU WiFi Deauther ESP8266 voor PS4-WIFI Starter Kit
Merknaam: lilygo (CN)
Aansluiting: Plug-in
Uitvoertype: DC
SKU: G510
Processor:L106 32-bit RISC microprocessor core based on Tensilica Xtensa
Diamant Standaard 106Micro running at 80 MHz
Memory:32 KiB instructie RAM
/32 KiB instruction cache RAM
/80 KiB user RAM
/16 KiB ETS system RAM
External QSPI Flash: 16 MiB is supported

**Attempts**
Also tried firmware release: esp8266_deauther_2.6.1_DSTIKE_NODEMCU_07_V2.bin

LeoeLeoeL, 2021-10-16:
2.6.1 doesn't work for me too.
Try 2.6.0. It works (the blue led turn on during deauthering)

polleke67, 2021-10-17:
The LED works indeed when downgrading to v.2.6.0, but still not de-authenticating connect requests.

LeoeLeoeL, 2021-10-17:
It works but, in order to test, you have to switch off your 5 GHz Wifi.

polleke67, 2021-10-17:
> 
> 
> It works but, in order to test, you have to switch off your 5 GHz Wifi.

LeoeLeoeL , I have to belief you on that, since I cannot deactivate 5Ghz on my ISP modem, not on my Samsung android phone....