SpacehuhnTech/esp8266_deauther #1496: Can only disconnect if my network is WEP.

Semiran, 2021-12-07:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

**Describe the error**

If is use wifi security wpa my devices don't disconnect if I use deather. When I only enable WEP in my router then deauther works and my devices disconnect.

What do you think causes it?

I don't know why the deauther don't work with wpa. I use a AZDelivery NodeMCU Amica Modul V2 ESP8266 ESP-12E WIFI Wifi Development Board with CP2102

captclearleft, 2021-12-13:
Do you have the router information (model number)?
Upload a screen shot of your phone wifi when you are running the deauth on the wpa.

closed 2021-12-13 by Semiran

captclearleft, 2021-12-13:
Interesting... Do you have a link?