SpacehuhnTech/esp8266_deauther #1497: Problems with web

mchyla7, 2021-12-08:
> Have you searched for existing (open and closed) issues describing the same problem?

No

**Describe the error**
The web isn't working and I don't know why. Back about 10 hours it works and now not

```
Error/Compile/Output Log
```

**To Reproduce**
Steps to reproduce the behaviour:
1) Connect to pwned
2) Open Opera and go to 192.168.4.1
3) Agree with terms
4) Press the button for scan and nothing
5) Try to go to settings and just text 

**Expected behaviour**
Normal working web with working buttons, settings and all that stuff 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: Android
 - Browser: Chrome, Opera 
 - Version: Latest
 - Hardware: **NodeMcu** CP2102 Lua WI-fi **ESP8266**

**Attempts**
I erase all settings and upload them again over Arduino software

**Additional context**
I try to power it through a power bank, and lately by PC USB 

edit:
Now its working and I don't know why :D
Now it's connected through the computer, through the same port but last time it didn't work anyway

Have anybody the same problem as me? And how did you solve it?

captclearleft, 2021-12-13:
It looks like you were using an Apple OS.  Most current OS's do not like a wifi connection that does not have internet access.
When you press scan, the esp8266 cannot "scan" and provide the web interface at the same time.  This shows the OS that the wifi is "disconnected" (in simple terms).  If a better wifi is available the Apple device (or any device) will connect to another known working website.  When you try to connect to 192.168.4.1 - It cant be accessed either because it is unavailable on the new wifi connection or the esp8266 is currently busy "sannning".  

You need to remove the known wifi's in that area and force the connection even though there is no internet connectivity.
Hope that helps.

P-A-Trick, 2021-12-14:
The same prob, if I connect via WLAN the connection is stable but the Webinterface fon"t reac5t...meas I can't see the settings or can do a scan via web...reset reload reboot nothing happens

captclearleft, 2021-12-14:
Just to clarify.  In the Wiki - It states:
1. The web interface is not available during scan.
2. Remove know wifi APs in your phone that it would normally connect to.

If using android you can force the wifi connection even if no interenet.

P-A-Trick, 2021-12-14:
I got the connection via WLAN to owned, but after accepting the rules the chapters  with menu scan attack settings and do on appears...
But the menus without function no reload reboot or reset is helping

captclearleft, 2021-12-15:
I don't understand what issue you are having.  Please provide a picture of your setup.  What device are you using?
 
I don't understand what you are saying here -

"..but the Webinterface fon"t reac5t...meas I can't see the settings or can do a scan via web..."   **Please provide a screen shot or video of what is not working.**  

And I don't understand what you are saying here - 

"but after accepting the rules the chapters with menu scan attack settings and do on appears..."
**Please provide a screen shot or video of what is not working.**

P-A-Trick, 2021-12-15:
[image]

P-A-Trick, 2021-12-15:
There are no options

P-A-Trick, 2021-12-15:
[image]
[image]

captclearleft, 2021-12-16:
@P-A-Trick The below image is what I am going off of.  If this is NOT what you have?  Create a new thread.

I am helping you with a NodeMCU esp8266 that you flashed with v2.6.1 as outlined in the tutorial.
Your images show that you have version 2.6.0.  I would update to the latest version.

[image]

P-A-Trick, 2021-12-16:
The version o  the V2 clock is 2.61... in the web it's 2.60 don't know why.
sorry I 'm new to this and also to the GitHub. Try to understand how
everything works here..

closed 2022-07-10 by stale[bot]