SpacehuhnTech/esp8266_deauther #1498: Lolin NodeMCU v3 (CH340) No Display, Resets

captclearleft, 2021-12-10:
I will continue to look for and post a solution. 

I am having the same exact issues described as resets constantly in the Testing section of the tutorial.  Following the "**Setup Display & Buttons**" exactly. (NodeMCU with Display Example I2c) 
I am using the LOLIN NodeMCU v3 with the ch340 chip (I have 3 of them, and have tried 2 of em).
I tried changing the code for the SDA and SCK pins from 5, 4 to 4,5 to D1,D2.  I tried swapping those wires.  
I tried uncommenting these lines in the A_config.h... 
       #define NODEMCU
       #define DISPLAY_EXAMPLE_I2C
I tried different power sources.
I tried just flashing it with the Node MCU Flasher and the .bin file.

I don't have any custom anything going on, just set up exactly as described in the tutorial.  

The regular NodeMCU code without display works flawlessly on both boards.  
When I run the OLED display code from https://github.com/squix78/esp8266-oled-ssd1306 The display works fine.  No resets, and the display shows everything just fine.   

I am guessing its something to do with the way this board and the ch340 chip exists.  
Maybe ill try the resistors on D4 and D8.

Any one else use this board, or have a solution?  Thanks so much.

_Originally posted by @captclearleft in https://github.com/SpacehuhnTech/esp8266_deauther/issues/1452#issuecomment-990300764_

captclearleft, 2021-12-10:
Just another update. Tried adding the resistors (D4 - High, D8 - Low, and I even tried Pulling Reset High (10k))
Same results - No display, constant resets...

wdt reset
load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000cb3d0
~ld

Reloaded the SSD1306SimpleDemo.ino sketch. Screen still works fine.
I still think its something screwy with this ch340 chipped LOLIN NodeMCU.

I did try my Adafruit feather esp8266 - That didnt work either. I may have not hooked that up correctly as there are only the two SDA and SCK (4,5) on that board and thats what I used...

I updated all my libraries for Arduino, still no luck. Any ideas would be appreciated. Thanks

captclearleft, 2021-12-10:
Tried changing SDA to 13 and SCK to 12.  That did not work - Same results.

captclearleft, 2021-12-10:
Ok, I tried adding capacitors to the power line and the 3.3v line.  No luck

captclearleft, 2021-12-11:
I think we got this solved with the help of @strolll395 (https://github.com/strolll395)

https://github.com/wh1ter0z/esp8266_deauther/blob/v1/esp8266_deauther/A_config.h

Ok, So I copied the above A_config.h into the Arduino IDE replacing the original A_config.h code.
Used "NodeMCU" As the "Board" and the "Deauther Config" under tools.
And it worked.
It looks like there is something in the #if #elseif statements when using the "Deauther Config= Deisplay Example IIC" mode that causes an issue. Using just the file with all the #if #elseif statements removed and then just defining the display worked.
Thanks
https://github.com/wh1ter0z/esp8266_deauther

closed 2021-12-12 by captclearleft

captclearleft, 2021-12-12:
I actually think the rests were being caused by the LED settings 
In the default A_config file these are commented out:

#define LED_DIGITAL
#define LED_PIN_R 16 // NodeMCU on-board LED
#define LED_PIN_B 2 // ESP-12 LED

I found that this being commented out was causing resets:  #define LED_DIGITAL
I now have it uncommented.

clisterX, 2022-10-25:
I was giving up after hours of trial and error and then in a last attempt I copied your code and it's working don't know why, don't know how. Thank you very much!