SpacehuhnTech/esp8266_deauther #1499: Can't Flash Nodemcu

VinhPham47, 2021-12-11:
I can't flash my nodemcu with esptool.py

It shows up as"A fatal error occurred: Invalid head of packet (0x80): Possible serial noise or corruption."

captclearleft, 2021-12-11:
Which model NodeMCU do you have ?
What operating system?
Do you have the correct drivers?
Have you had any success with writing/flashing the board - or is this your first attempt?

Check that you have the correct USB drivers?
Look at the boards pins, and see if there is solder overlap on any of the pins (Specifically the tx and rx pins)

idz3n, 2022-01-16:
I cant flash my Nodemcu Cp2102 which bin should i be flashing.. the correct com port was selected. I i verified by unplug the usb cable. please advise

closed 2022-09-24 by stale[bot]