SpacehuhnTech/esp8266_deauther #1511: Deauth and Beacon not working properly together on some versions [NodeMCU]

gcapris90, 2022-01-09:
Hello! First of all, I would like to thank you for sharing this project! It is a great tool for learning.

I've been using the version 2.5.0 RC2 for quite a long time and just now decided to update it to the version 2.6.1 (latest so far).

After following the steps and successfully compiling and uploading using Arduino IDE, I've noticed that the version 2.6.1 won't function as the version I was using before. When trying to turn both **Deauth and Beacon** attacks at the same time, the "Stop" button becomes "Start" and the led on my board turns off. The same thing occurs with the versions: 2.6.0; 2.6.0 RC3 and 2.6.0 RC2. Other than that, the led while using version 2.6.1 is **not turning on** at any moment on my board.

Although not so stable, the version 2.6.0 RC1 seems to be working fine when trying to perform two attacks at the same time. The "Stop" button action is often quite delayed when clicking on it. I will keep on testing it.

I'm using a NodeMCU (ESP8266MOD) with Arduino IDE the libs all updated. I've tested it on the PC using both Windows 7 and Ubuntu 20.04.01 LTS  and using Chrome (Version 97.0.4692.71) and Brave (Version 1.34.80 based on the same Chromium version mentioned). Also tested it on Mobile using Chrome (Version 97.0.4692.70) both on a Xiaomi MI 9T Pro and on a Samsung Galaxy S5. 

I don't know if I'm doing something wrong or if maybe this is in fact some sort of bug.

closed 2022-09-24 by stale[bot]

SuthaharanTharunjan, 2024-08-16:
Me having this error

varya33, 2024-12-07:
Have you figured out how to solve this problem?

NguyenDucBaoMinh, 2025-03-30:
If you are experiencing this, please revert to version 2.5.0 RC2.