SpacehuhnTech/esp8266_deauther #1521: Does this DEAUTH 5GHz ?

tanitoo2, 2022-02-01:
when i use it on my network it only deauth 2.4ghz wifi not 5GHz, i am correct ? or it works on 5ghz?

m4rch1n0, 2022-02-01:
works only on 2.4GHz

tanitoo2, 2022-02-01:
wow, today almost every wifi runs 5GHz, is there any way to come around this issue ?

m4rch1n0, 2022-02-01:
It’s not an issue it’s just that the esp8266 network card can reach only 2.4GHz networks

tanitoo2, 2022-02-01:
oh, ok got it, thanks, also i flashed my esp8266 card with the oldest bin, what BIN file should i use, thanks !
esp8266_deauther_2.6.1_AVATAR_5W_E14_LAMP.bin815 KB
 esp8266_deauther_2.6.1_DISPLAY_EXAMPLE_I2C.bin815 KB
 esp8266_deauther_2.6.1_DISPLAY_EXAMPLE_SPI.bin814 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_BOY.bin815 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_MINI.bin816 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_MINI_EVO.bin817 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_MOSTER.bin815 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_MOSTER_V2.bin815 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_MOSTER_V3.bin815 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_MOSTER_V4.bin815 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_MOSTER_V5.bin817 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_OLED.bin815 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_OLED_V1_5.bin815 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_OLED_V1_5_S.bin815 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_OLED_V2.bin814 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_OLED_V2_5.bin814 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_OLED_V3.bin814 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_OLED_V3_5.bin815 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_OLED_V4.bin815 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_OLED_V5.bin815 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_OLED_V6.bin817 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_V1.bin814 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_V2.bin814 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_V3.bin814 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_V3_5.bin815 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_WATCH.bin816 KB
 esp8266_deauther_2.6.1_DSTIKE_DEAUTHER_WATCH_V2.bin817 KB
 esp8266_deauther_2.6.1_DSTIKE_D_DUINO_B_V5_LED_RING.bin815 KB
 esp8266_deauther_2.6.1_DSTIKE_NODEMCU_07.bin814 KB
 esp8266_deauther_2.6.1_DSTIKE_NODEMCU_07_V2.bin815 KB
 esp8266_deauther_2.6.1_DSTIKE_USB_DEAUTHER.bin814 KB
 esp8266_deauther_2.6.1_DSTIKE_USB_DEAUTHER_V2.bin815 KB
 esp8266_deauther_2.6.1_HACKHELD_VEGA.bin815 KB
 esp8266_deauther_2.6.1_LYASI_7W_E27_LAMP.bin815 KB
 esp8266_deauther_2.6.1_MALTRONICS.bin817 KB
 esp8266_deauther_2.6.1_NODEMCU.bin814 KB
 esp8266_deauther_2.6.1_WEMOS_D1_MINI.bin

m4rch1n0, 2022-02-01:
It depends on which board you are using, almost every board has an ESP8266 chip

tanitoo2, 2022-02-01:
ok, when you scan for APS, how do you know wich is 2.4 and 5, how do you know u are running an attack that is actually working, because you dont have a feedback

tanitoo2, 2022-02-01:
if i run the attack in my home network, i know its working because it kicks me from the network 2.4ghz, but when i attack my 5ghz it does nothing, so my question is, if you run an attack how do you know its a 5ghz or 2.4ghz

m4rch1n0, 2022-02-01:
5GHz networks will not be shown as the card does support them up at all. Remember that most routers are dual band, which it means that they both have 2.4GHz and 5GHz, but if someone is connected to the 5GHz network nothing will happen. Btw if a device supports only 2.4GHz band or maybe the device is too far to get 5GHz band (5GHz covers less than 2.4GHz) it will be affected by the attack

tanitoo2, 2022-02-01:
yes, but the thing is that i DO see 5GHz networks on the device, thats why im asking and i can corroborate that with airodump -ng, i see both of my networks in the ESP8266 2.4GHz and 5GHz

m4rch1n0, 2022-02-01:
May I ask which board are you using… I’ve just tried to split my network in two different networks (2.4 and 5) and I actually saw that it doesn’t show 5GHz network, but I was using a monster oled board so I can see networks directy from there… maybe if you are connected through a device that has a network card that supports 5GHz networks it shows them to you… otherwise i can’t explain

tanitoo2, 2022-02-01:
im using this esp8266_deauther_2.6.1_AVATAR_5W_E14_LAMP.bin815 KB, the networks that are in 5Ghz are shown as ''HIDDEN'' i cant attack it but nothing happens

closed 2022-09-24 by stale[bot]