SpacehuhnTech/esp8266_deauther #1522: need urgent help

toxic-dew, 2022-02-02:
the problem is that I'm trying to catch my phone thief. I've got the location but don't know the height as it is a multi-floored apartment building. I have to give the police the right location because it was me who tracked the thief. I have the phone's info including the mac address of wifi (thank god for giving me the idea to mac filter my home network). I was wondering if the dormant two leads of my nod mcu can be programmed to strobe according to the signal strength of the wifi if the phone is connected to the wifi. I meant to say it will be looking for the wifi with that mac address ......when it does it will start flickering to the signal strength. I was thinking of trying this idea myself, but I'm still new to programming, all I know is basic c/c++ and a little bit of python.
could you introduce this feature or give some idea on how to do it,please?
thanks in advance
[image]

Kyr4l, 2023-03-21:
this isn't the right place for this topic, not an issue related to the firmware

Kyr4l, 2023-03-21:
> Ok come to river app...and Don't enter dark web
> […](#)

cringe