SpacehuhnTech/esp8266_deauther #1526: Port tab under tools is greyed out, but the esp8266 is plugged into the computer

Zach5678, 2022-02-15:
The port tab is greyed out as shown in the picture and I can not click it and select a port.
[image]

I have the esp8266 plugged into the computer, and when I plugged it in, the esp8266 blinked, but did not do the windows noise thing when it gets plugged in. 

I have done everything else up to thing point correct I think, but it can not upload it because of me not being able to select the port Any help would be greatly appreciated since I have been trying to figure this out for a few hours lol.

spacehuhn, 2022-02-15:
If you don't see a new port appear when you plug in your board, or the menu is grayed out completely, then no new serial port was detected.  
This can be caused by a missing driver or a defective USB cable. 

Learn more: https://blog.spacehuhn.com/espcomm/

Zach5678, 2022-02-15:
Ok, I looked at the link you sent and I downloaded the driver for my board for the problem, but what do I do with the downloaded file after that? Thanks for helping

Zach5678, 2022-02-15:
Ok I know I keep doing comments but I think I found the problem the thing I was using to connect the board to the computer wasn't an adapter but just a regular charger for like a Samsung or something. I'm going to get the actual adapter and hopefully the problem will be solved but thanks for your help so far.

Zach5678, 2022-02-16:
Ok, I finnaly got it to work. Just in case other people had this problem i want to say what i did to fix it. I first went to the link https://blog.spacehuhn.com/espcomm/ and downloaded the driver. I extracted the file and went into it and went to the setup tab downloaded that. I then went to device manager, and pluged in the board. the usb thing popped up and i clicked update when i clicked the usb. then i changed the driver to what i just downloaded and it worked. just in case someone wanted to know what i did to fix it. Thanks

closed 2022-02-16 by Zach5678