SpacehuhnTech/esp8266_deauther #1531: Bootloop when flashing Nodemcu with Display Example IC2

danizwam, 2022-03-18:
Hello,

I flashed version 2.1 on my Nodemcu some time ago. 
I soldered an LCD and a few buttons onto a circuit board and connected it to the Nodemcu. This worked perfectly. In addition to the self-build, I bought the Watch V2.

Yesterday I wanted to update both to 2.61. The Watch V2 was updated flawlessly, but I have problems with the Nodemcu. If I flash without "Display Example IC2", the Nodemcu starts without problems. The "pwnd" WLAN is created and I can connect. As soon as I select "Display Example IC2 / SPI", I end up in a bootloop. This happens both via the Arudino IDE and with the BIN files from Github.

If one connects to the serial monitor, the following block repeats itself´:

` ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

Mounting SPIFFS...OK
Loading settings...OK
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: false
STARTED! \o/
2.6.1
`

I have no idea what else I can try.
Does anyone have any ideas?

Thank you very much in advance.

Greetings, Daniel

Rapa2k, 2022-03-20:
I have the same issue with the Display Example IC2 bootloop, if i flash with NodeMCU example a dont have the buttons working

CoulsTech, 2022-03-22:
You have to change the device target type to NodeMCU, as per the installation guide, 
https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide
Shows how to add the SpaceHun Deauther board choices and how to select them

danizwam, 2022-03-22:
> You have to change the device target type to NodeMCU, as per the installation guide, https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide Shows how to add the SpaceHun Deauther board choices and how to select them

Hi,

thank you very much for your contribution.
I have already tried various settings, but it just doesn't work as desired.

These are my settings:

Tools -->:
--> Board --> Deauther ESP8266 Boards (2.7.5)  --> NodeMCU
--> Upload speed: 115200
--> Flash Size "4MB (FS:1MB OTA:~1019KB)"
--> Erase Flash: "All Flash Contents"
--> Deauther Config: "Display Example I2C"

This causes the bootloop.

If I change "Deauther Config": to "NodeMCU", i.e. without display, the NodeMCU boots correctly and the "pwnd" network is displayed, but neither the display nor the buttons work.

If I change the settings to ESP8266, as you suggested, and configure all the parameters by hand, it cannot compile.

Kazem-ma79, 2022-04-12:
Hi
I'm having the same problem but found the fix.
Set Deauther Config to NodeMCU and for using OLED just paste code below to your A_Config.h (I recommend pasting before commented defines).

// Display
#define SSD1306_I2C
#define I2C_ADDR 0x3C
#define I2C_SDA 4      // D2
#define I2C_SCL  5      // D1

danizwam, 2022-04-13:
> Hi I'm having the same problem but found the fix. Set Deauther Config to NodeMCU and for using OLED just paste code below to your A_Config.h (I recommend pasting before commented defines).
> 
> // Display #define SSD1306_I2C #define I2C_ADDR 0x3C #define I2C_SDA 4 // D2 #define I2C_SCL 5 // D1

Hi, thanks for pointing me to the right file.

Unfortunately your config does not work for me,  but some changes did the job ;)

My IC2 ist SDA 5/SCL 4  and yours is 4/5 wired and i had to remap my three buttons:

This is my config:

  #define USE_DISPLAY true
  #define FLIP_DIPLAY true

  #define SSD1306_I2C
  #define I2C_ADDR 0x3C
  #define I2C_SDA 5      // D2
  #define I2C_SCL 4      // D1

  #define BUTTON_UP 14
  #define BUTTON_DOWN 13
  #define BUTTON_A 12

I pasted it in the A_Config.h right before the commented defines, as suggested!

Thank you VERY MUCH for you help!

closed 2022-04-13 by danizwam

ASCII-encrypt, 2023-02-25:
> Hi I'm having the same problem but found the fix. Set Deauther Config to NodeMCU and for using OLED just paste code below to your A_Config.h (I recommend pasting before commented defines).
> 
> // Display #define SSD1306_I2C #define I2C_ADDR 0x3C #define I2C_SDA 4 // D2 #define I2C_SCL 5 // D1

ive tried this and i still get a bootloop and now theres bars on the display

elimerins, 2023-07-09:
> Hi, thanks for pointing me to the right file.
>
> Unfortunately your config does not work for me, but some changes did the job ;)
>
> My IC2 ist SDA 5/SCL 4 and yours is 4/5 wired and i had to remap my [...]

Such a good help. 

but in my case, complete compile. still i CANT see my dispaly. i need check.

elimerins, 2023-07-10:
> Hi, thanks for pointing me to the right file.
>
> Unfortunately your config does not work for me, but some changes did the job ;)
>
> My IC2 ist SDA 5/SCL 4 and yours is 4/5 wired and i had to remap my [...]

very helpful my display show informations.

moeburn, 2024-06-03:
> Hi, thanks for pointing me to the right file.
>
> Unfortunately your config does not work for me, but some changes did the job ;)
>
> My IC2 ist SDA 5/SCL 4 and yours is 4/5 wired and i had to remap my [...]

Didn't work for me.  Can't get this software to stop bootlooping on my ESP8266.  Doesn't matter that I've got the correct NodeMCU board selected, doesn't matter what I add into the a_config file.