SpacehuhnTech/esp8266_deauther #1539: how can i disable deauther ap?

1999AZZAR, 2022-04-15:
anyone can help how do i can disable the default pwnet access point? cause i am using a diy device with an oled and button, and i have a plan to make it power efficient by reducing the oled standby time (i have done that) also i remove all the led on my nodemcu board, but i just need one more thing that is how to disable the default wifi ap when i compile it using arduino ide?

Timmotools, 2022-08-12:
connect serial cable and use the commands via a serial terminal: 
set webinterface false
save settings
reboot

you might be able to disable it. basically when you disable the web interface it will disable the pwned AP.

1999AZZAR, 2022-08-14:
I have disable it via pre compiled file but it's (pwnet AP) still there after i compile it, but I'll consider trying your suggestion

Timmotools, 2022-08-14:
I just used the precompiled binaries. As I use a nodeMCU via serial and with my FlipperZero I had no need to use the AP. I did start looking though the source and came across the setting to disable the web interface. using the serial command you can use "get settings" and it will show the options. very handy. try it and you will be surprised.

closed 2023-04-02 by stale[bot]