SpacehuhnTech/esp8266_deauther #1544: Not working display

D1st3f, 2022-04-28:
Hi, i have esp8266 and sdi oled display. I conected like wiki, but nothing work. i tried flash by Arduino IDE and by bin file. If i flash Nodems version all works, but if i flash SPI version anythink dont work, no wifi to connect and no image. 

[image]
[image]
[image]

I tried display demo and display work fine. 
Pls help : /
Thxs

D1st3f, 2022-04-29:
I fixed that sht : ) In wiki you have issue, ssd1306_Spi you have double usage of same pin, same issue you have in bin file. Pls fix that. I remove double define pin and all works nice : ) 

#elif defined(DISPLAY_EXAMPLE_SPI)
   #define SSD1306_SPI

  **_#define SPI_RES 5_**
  #define SPI_DC 4
  #define SPI_CS 15

 #define FLIP_DIPLAY true

// ===== BUTTONS ===== //
  #define BUTTON_UP 0
  **_#define BUTTON_DOWN 12_**
  #define BUTTON_A 2

// ===== LED ===== //
  #define LED_NEOPIXEL_GRB
// #define LED_NEOPIXEL_RGB

  #define LED_NUM 1
  #define LED_NEOPIXEL_PIN 9
  #define LED_MODE_BRIGHTNESS 10

#elif defined(MALTRONICS)

// ===== Reset ====== //
  **_#define RESET_BUTTON 5_**

// ===== LED ===== //
  #define LED_DOTSTAR
  #define LED_NUM 1
  _**#define LED_DOTSTAR_CLK 12**_
  #define LED_DOTSTAR_DATA 13
  #define LED_MODE_BRIGHTNESS 255

jakobkarosseit, 2023-05-05:
What do i have to change?