SpacehuhnTech/esp8266_deauther #1548: Fetal error in v3

PPCVRJCUV, 2022-05-04:
In file included from G:\esp8266_deauther-3\esp8266_deauther\esp8266_deauther.ino:7:0:
debug.h:11:10: fatal error: climits: No such file or directory
 #include <climits>   // LONG_MAX
          ^~~~~~~~~
compilation terminated.
exit status 1
climits: No such file or directory
 
I don't even see climits header file  .... ! 
the error occurs in debug.h file  , so please checkout that

captclearleft, 2022-10-18:
NOTE: Version 3 .
I downgraded the Deauther ESP8266 Boards Library to 2.7.4 and the code compiles and loads to the board. WITH NO ERROR.
It seems that the 2.7.5 Library has the error.  as if right now - thats what I have so far.

captclearleft, 2022-10-18:
SOLVED WITH Downgrade. Downgrading the Deauther ESP8266 Boards Library to 2.7.4 fixed the issue for me.  
Can anyone confirm this.?

hcaga, 2022-11-15:
Yes, this is confirmed. I downgraded the ESP8266 library to version 2.7.4 and it fixed the problem.

ra1n08, 2023-01-15:
bruh but i need feature in V3 :v

closed 2023-09-17 by stale[bot]