SpacehuhnTech/esp8266_deauther #1552: ability to work with flipper zero wifi dev board

luigi20000, 2022-05-15:
hello would it be possible to make this code work with flipper zero wifi dev board myself i am not yet strong enough in programming to work this in

deft01, 2022-07-13:
up

hsiboy, 2022-09-11:
I'm just about to clone the src and compile for the Flipper WiFi Dev board (ESP32)

deft01, 2022-09-13:
Nice, the problem is that the flipper zero dev board doesn't seems to support at all 5 Ghz bandwidth. Loosing a lot of it's interest imo.

hsiboy, 2022-09-13:
The ESP8266 (that this project uses) doesn't offer 5ghz either, so I'm not sure what your comment is referring to?

deft01, 2022-09-13:
Didn't dig enought in the ESP8266 deauther project details (this project) to found out it wasn't supporting 5ghz either. Do you have in mind a similar deauth project supporting the 5ghz bandwidth ?

deft01, 2022-09-13:
I could be really interested buying it