SpacehuhnTech/esp8266_deauther #1561: Noob to arduino  !  Cant get display running...

Roosterno1, 2022-06-11:
Totally new to all this, so no "techy" answers please !
I have succesfully installed deauth onto a ESP8266  wemos d1-mini, this has worked for months using 2.6 code...
So I decided to experiment...
Ive managed to hook up an OLED display 1306 (?)  to another arduino clone board, and it works fine , but when I connect it to the d1-mini it just sits there blank  !!
I am sure the connections are ok, and the reflashing goes fine - I can see the output in arduino seriel output...
Its the configuring the sketch im having problems with... I think im supposed to alter something in the "A_config.h" tab ?

The display works on address 0x3c 

Using arduino ide on a windows 10 pc... im sure the port config is ok, as it DOES program the D1, and I can connecto to it via wifi on my phone, so the code is running.

Please can you help ?

KAJPER, 2022-08-18:
Can you help me with problem?
pls can you explain what is reversed

Budiono55, 2023-01-12:
> Can you help me with problem? pls can you explain what is reversed

based on @Roosterno1 problem, it looks like the SDA and SCL connection were reversed. You can see the board pinout to find out more where is your pins.

closed 2023-09-17 by stale[bot]