SpacehuhnTech/esp8266_deauther #1570: Beacon

Dali1305, 2022-08-04:
Hello I wanted to use the esp8266 deauther to do beacon attack but when I launch the attack there is nothing in my wifi list I waited some time but still nothing can someone help my please

icdmkg2, 2022-09-14:
i have the same issue and still didnt find a fix for it

mastersubhajit, 2022-09-20:
# Assuming that your ESP8266 Deauther ver 2.6.0 is flashed correctly, follow the following steps : 

* For Customized Beacon Attack-
1. Visit SSIDs section and input a SSID name, 
2. choose security type and number of fake SSID wanted
3. Turn on Overwrite 
4. Click on Add
5. Go to Attacks section 
6. Start Beacon Attack.

* For Random SSID Beacon Attack-
1. Visit SSIDs Section
2. Enable Random Mode
3. Reload once 
4. Got to Attacks Section
5. Start Beacon Attack 

I hope your issue is resolved by now if not watch this youtube video by spacehuhn -https://www.youtube.com/watch?v=TMvpJiE8Yy8

Dali1305, 2022-09-20:
> 

Do you think that I did not do that ?

mastersubhajit, 2022-09-21:
> > 
> 
> Do you think that I did not do that ?

Then there must be some problem with your installation, please do it one more if possible.

closed 2023-04-02 by stale[bot]