SpacehuhnTech/esp8266_deauther #1573: Buttons not working

to6koto6ko, 2022-08-15:
I tried to install the deauther 2.6.1 on few esp8266 devices. Unfortunately the buttons are not operational at all. There are two Nodemcu 8266 v1 where  the display (I2C) and everything else is working properly except the buttons. I have one more ESP8266 noname and the result is the same.
the config is default:
// ===== BUTTONS ===== //
  #define BUTTON_UP 14   // D5
  #define BUTTON_DOWN 12 // D6
  #define BUTTON_A 2     // D4
  #define BUTTON_B 0     // D3
but connecting to the GND is without any change.  
The terminal command : screen mode buttontest
shows only false on all 4 buttons.
there is no difference between  D5,D6 or D3 connected to the GND or not

fcamargo10, 2022-10-01:
Same here! How you fix ?

goliath29, 2023-04-28:
Hi, has there been a solution for this issue? Experiencing the same myself

sinaun4given, 2023-04-28:
i find that.u must use dstike bin file .in this version button works.bro
you gonna love me when try

goliath29, 2023-04-28:
> i find that.u must use dstike bin file .in this version button works.bro you gonna love me when try
> […](#)

You god damn legend. I love you.
The DSTIKE deauther .bin worked perfectly!!!
Thank you so much

sinaun4given, 2023-04-30:
your welcome😍😍😍😍

glen4cindy, 2023-06-03:
I've got the same problem. I've tried 3 different dev boards and all 3 are the same. The screen comes on and the web interface works. I've tried using the alternate bin file and I've tried flashing with Arduino IDE and ESPWebTool.

I've also made sure to uncomment the appropriate lines referring to buttons D5 and D6 when I used Arduino IDE. Still no joy when it comes to using the buttons.

I've even tried using the wires directly on the pins, D5 to GND, D6 to GND without success.

Any suggestions would be appreciated.

yashgunjal04, 2023-06-14:
same bug I also faced how to fix it ?

I tried to install the deauther 2.6.1 on few esp8266 devices. Unfortunately the buttons are not operational at all. There are two Nodemcu 8266 v1 where the display (I2C) and everything else is working properly except the buttons. I have one more ESP8266 noname and the result is the same.
the config is default:
// ===== BUTTONS ===== //
#define BUTTON_UP 14 // D5
#define BUTTON_DOWN 12 // D6
#define BUTTON_A 2 // D4
#define BUTTON_B 0 // D3
but connecting to the GND is without any change.
The terminal command : screen mode buttontest
shows only false on all 4 buttons.
there is no difference between D5,D6 or D3 connected to the GND or not

sinaun4given, 2023-06-14:
download dstike version . button works in this version.you gonna love me
bro after test

glen4cindy, 2023-06-15:
> same bug I also faced how to fix it ?
>
> I tried to install the deauther 2.6.1 on few esp8266 devices. Unfortunately the buttons are not operational at all. There are two Nodemcu 8266 v1 where the [...]

Same issue here.

I've tried about 5 different boards now.  I found ONE where the buttons work. Oddly enough it's a board with an integrated display but the integrated display doesn't work. I just realized it uses different pins than D1 and D2.

I tried the same boards with another sketch wired to buttons and LED's using the same pins. They all worked.

I'm thinking there is another issue. Maybe it's some combination of what board and configuration you select or something. I'm so very confused.
 
I tried the suggestion of dstrike and unless I'm doing something wrong that didn't work either. 

I'm not sure what to try next.

yashgunjal04, 2023-06-15:
Which version , theay have lot of versions available
please help bro

glen4cindy, 2023-06-15:
> Which version , theay have lot of versions available please help bro
>

I was finally able to get one of my boards working.

The D1 mini is currently working by using the .bin version.

My previous attempts were done using the .zip versions and the Arduino IDE software to compile and write the file to the board.

billypolidorou, 2023-09-06:
same problem there :( i need help :) pls help me pls !!

MaeseppTarvo, 2023-10-28:
Has someone figured out the button problem?
I myself went though the code and can not find the specific place where the bug is. I found out that the A button is always "PRESSED". So I disconnected it. Now I have UP and DOWN. I was able to activate the button press if I use 10kohm resistor and touch ground with it. I also found out that if you use ESP8266 esp-12 0 pin for button. This is the only button that works 😄

ArticPolarDEV, 2023-11-01:
> same problem there :( i need help :) pls help me pls !!

Same problem, bro. I need Help!

CyberTechNex, 2024-04-26:
> download dstike version . button works in this version.you gonna love me bro after test
> […](#)

which version of the dstike should i use i have a node mcu and trying to get the buttons working with the ssd1306 display

SkyNezu, 2024-06-16:
I have a dstike deauther monster v5. The up-down-OK button does not work. The other buttons work. What firmware version do I need to use to make all the buttons work?

sinaun4given, 2024-06-16:
can you tell me whats the gpio of others button in your version?we are
talking about nodemcu esp ,sorry idont know about your version(v5 ...) if
you want to make this with nodemcu i can help you

SkyNezu, 2024-06-17:
> can you tell me whats the gpio of others button in your version?we are talking about nodemcu esp ,sorry idont know about your version(v5 ...) if you want to make this with nodemcu i can help you
> […](#)

I made a slight mistake. I have this device, link: https://dstike.com/products/dstike-deauther-monster-v2

All buttons work. Except for the up-down-ok button. I used the firmware for monster (bin - dstike - monster) on the page https://deauther.com/docs/download. I installed the firmware through the esp.huhn.me website. The firmware went without errors.

tihnar228, 2024-07-03:
Hi guys. have you figured out how to make the buttons work? Everything worked fine before but now it doesn't work. I tried many firmwares.

scienceLabwork, 2024-08-08:
There is a small correction in the code
1. Go to A_config.h
2. Scroll down to line-**378** or until you find following lines
`#elif defined(DEFAULT_ESP8266) || defined(NODEMCU) || defined(WEMOS_D1_MINI) || defined(DSTIKE_USB_DEAUTHER) || defined(DSTIKE_NODEMCU_07) || defined(DSTIKE_DEAUTHER_V1) || defined(DSTIKE_DEAUTHER_V2) || defined(DSTIKE_DEAUTHER_V3)`
3. below that add following lines
`
  #define BUTTON_UP 12
  #define BUTTON_DOWN 13
  #define BUTTON_A 14
`

Hope, This works for you.

ghosthets, 2024-09-25:
same in my esp8266 i setup and succesfuly all thing are work properly like display and esp but the buttons are not working whyy,, plz give me solution

ghosthets, 2024-09-25:
> same problem there :( i need help :) pls help me pls !!

plz say u r problem solve or not????

Vipulraj0152, 2024-10-10:
Same issue i have changed 4 boards... it seems working perfect on d1 mini..

ATLNIKHIL, 2025-01-13:
me to friends

iprosenjit9, 2025-05-10:
From line 574 to 584 change 255 to respective button pin value
-------From this-----------
#ifndef BUTTON_UP
#define BUTTON_UP 255
#endif // ifndef BUTTON_UP

#ifndef BUTTON_DOWN
#define BUTTON_DOWN 255
#endif // ifndef BUTTON_DOWN

#ifndef BUTTON_A
#define BUTTON_A 255
#endif // ifndef BUTTON_A

--------To this------

#ifndef BUTTON_UP
#define BUTTON_UP 14
#endif // ifndef BUTTON_UP

#ifndef BUTTON_DOWN
#define BUTTON_DOWN 12
#endif // ifndef BUTTON_DOWN

#ifndef BUTTON_A
#define BUTTON_A 13
#endif // ifndef BUTTON_A