SpacehuhnTech/esp8266_deauther #1581: API

ra1n08, 2022-09-08:
does this esp8266_deauther project have support api?
And what communication does it support? (UDP, TCP/IP, Websocket, http)
i plan to make a deauther control app via esp8266_deauther api, unlike similar apps on play store, they all use http GET to esp8266 website

if there is no api hope the development team make more api to make the control easier :)

sorry for my bad english

translate Vietnamese to English by google translate :))

NurTasin, 2022-12-08:
I don't think that there is an API to control the behavior of the deauther.
However you can use `/names.json` ,`/scan.json`,`/ssids.json`,`/attack.json` endpoints to get some useful data from the
deauther. And you can send CLI commands to the deauther using the `/run` endpoint with the argument `cmd` that will contain 
[CLI commands](https://github.com/SpacehuhnTech/esp8266_deauther/blob/v2/serialcommands.md) for the deauther.
To know more about the json files and which variable means what you have to read through their code.

ra1n08, 2022-12-10:
> I don't think that there is an API to control the behavior of the deauther. However you can use `/names.json` ,`/scan.json`,`/ssids.json`,`/attack.json` endpoints to get some useful data from the [...]

i know but i want send commands via wifi, not serial monitor

closed 2023-06-17 by stale[bot]