SpacehuhnTech/esp8266_deauther #1596: ArduinoJSON v5 error

markdonofrio, 2022-10-26:
I installed ArduinoJSON v5 as per instructions but still get the error below when trying to verify / load the code.

Arduino: 1.8.12 (Windows 10), Board: "Generic ESP8266, NodeMCU, dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, 160 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), Disabled, None, v2 Lower Memory, All Flash Contents, 115200"

esp8266_deauther:23:2: error: #error Please upgrade/downgrade ArduinoJSON library to version 5!

 #error Please upgrade/downgrade ArduinoJSON library to version 5!

  ^

In file included from sketch\Settings.h:11:0,

                 from C:\Users\Gram\Desktop\Inventions\NRF Wifi Jam\esp8266_deauther-master\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:29:

A_config.h:7:24: fatal error: SH1106Wire.h: No such file or directory

 #include "SH1106Wire.h"

                        ^

compilation terminated.

exit status 1
#error Please upgrade/downgrade ArduinoJSON library to version 5!

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

q1w17, 2022-12-01:
Were you able to fix it? I also have the same problem

markdonofrio, 2022-12-02:
> Were you able to fix it? I also have the same problem

I used the old version of Arduino and it worked.  The new IDE version wouldn't work.

q1w17, 2022-12-02:
> > Were you able to fix it? I also have the same problem
> 
> I used the old version of Arduino and it worked. The new IDE version wouldn't work. It's much easier to get a raspberry pi and load it with the latest Kali pi ARM image. It can do 2.5 and 5g deauther as well as many other tools.

Hey there! Thx for the help. I tried with the 1.8.19 version of arduinoIDE and it didn't work. Can you tell me what version worked for you? Thx :)

closed 2023-06-17 by stale[bot]