SpacehuhnTech/esp8266_deauther #1600: UI not displaying

evanwheelrr, 2022-11-06:
I have searched for previous problems on here but couldn't find anything. I am running
 - OS: Windows 11
 - Browser: microsoft edge
 - Version: NA
 - Hardware DSTIKE

My problem is after I successfully flashed the firmware onto my esp8266 DSTIKE WIFI Deauth Detector It did not show the UI or interface I could use to control it. I got no errors when flashing it and need a explanation on how to display the interface

XGerman009, 2022-11-07:
Same, but I can see a "broken pixels", if check on other boards, oled display working fine! Using 2.6.0

evanwheelrr, 2022-11-07:
> Same, but I can see a "broken pixels", if check on other boards, oled display working fine! Using 2.6.0

you just need to use the boared called “deauther esp8266” i’d be hello to hello you on discord my discord:  evan.#5196

evanwheelrr, 2022-11-27:
I had not flashed it right or at least the flash failed at first

closed 2022-11-27 by evanwheelrr