SpacehuhnTech/esp8266_deauther #1606: 5G deauther

markdonofrio, 2022-11-19:
Is there a cheaper board I can use for the 5G deauther instead of the raspberry pi 3B+?

AnduinBrian, 2022-11-21:
i believe we dont have any hardware to support 5ghz at this time

markdonofrio, 2022-11-21:
In the video you use raspberry pi 3b+ is there a cheaper raspberry pi model I can use that will work the same way?

EPiCxRIOTZZ, 2022-11-26:
> i believe we dont have any hardware to support 5ghz at this time

the rtl8720dn supports dualband 5ghz but this project would need rewriting to support the RTL chip

behind24proxies, 2022-12-03:
> > i believe we dont have any hardware to support 5ghz at this time
> 
> the rtl8720dn supports dualband 5ghz but this project would need rewriting to support the RTL chip

ok rewrite it then

salmanfarisvp, 2022-12-06:
There are mainly three types of RTL/ Ameba boards. 
- [AMB21/AMB22](https://amebaiotdocuments.readthedocs.io/en/latest/ambd_arduino/AMB21/index.html)
- [AMB23 (RTL8722DM MINI)](https://amebaiotdocuments.readthedocs.io/en/latest/ambd_arduino/AMB23/index.html#amb23-rtl8722dm-mini)
- [BW16 (RTL8720DN)](https://amebaiotdocuments.readthedocs.io/en/latest/ambd_arduino/BW16_/index.html#bw16-rtl8720dn)
- [SparkFun AzureWave Thing Plus - AW-CU488](https://www.sparkfun.com/products/19957)

and all of them support Arduino SDK https://github.com/ambiot/ambd_arduino.

TTN-, 2023-02-13:
ESP32-C5 model has been released. Supports both 5.8ghz and 2.4GHz.
http://www.esp32learning.com/hardware/espressif-unveils-the-esp32-c5-a-dual-band-wi-fi-6-and-ble-risc-v-soc.php

closed 2023-09-17 by stale[bot]