SpacehuhnTech/esp8266_deauther #1610: Web-Console for v2

WaGi-Coding, 2022-12-08:
**Do similar feature requests issues already exist (open and closed)?**
Didn't found any

**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I like the Webinterface, but I like using CLI even more. I am not sure if it's possible already, but i'd like if i could use the AP for CLI.
Basically, serving a Website offering a output-textbox, command-textbox, send button and some configurable macro buttons. Also i am not sure if i am the only one who would use it.

**Describe alternatives you've considered**
Instead of adding it to the existing Web-Interface, offer a branch where instead of the current Web interface, it uses the Web-Console described above

**Additional context**
Here a screenshot of my current Serial Monitor to visualize the features a bit
[image]

NurTasin, 2022-12-08:
I found these pieces of code in `wifi.cpp`
```c++
server.on("/run", HTTP_GET, []() {
    server.send(200, str(W_TXT), str(W_OK).c_str());
    String input = server.arg("cmd");
    cli.exec(input);
 });
```
this takes CLI input as `cmd` argument on `/run` path. Something like `/run?cmd=help`.
I don't know I could be totally wrong. But it doesn't send the CLI output back.

WaGi-Coding, 2022-12-08:
> I'll test if it works by editing a line in the autostart file or so.

Okay, it works. You can see the output over serial monitor tho. If you want to send commands with spaces, use %20 instead of space

NurTasin, 2022-12-08:
> > I'll test if it works by editing a line in the autostart file or so.
> 
> Okay, it works. You can see the output over serial monitor tho. If you want to send commands with spaces, use %20 instead of space, for example `192.168.4.1/run?cmd=replace%20autostart.txt%200%20help` would replace the first line (0) of the autostart.txt (by default sysinfo) with the "help" command
> 
> But for sure you do not get the output as response unfortunately, which then is not very helpful in that matter :/ Anyway, very good to know

Yeah, argument values must be URL escaped. Thats why we have to use %20 instead of space.
I think I can make some progress on sending the cli output as response. I am currently reading the [`cli.cpp`](https://github.com/SpacehuhnTech/esp8266_deauther/blob/v2/esp8266_deauther/CLI.cpp).  There I've found the `void CLI::runCommand(String input)` function. I was wondering if somehow we override the `prntln` and `prnt` function so that they print on Serial monitor as well as sending the texts as a response to the web console, we will be able to achieve our goal. I'll try to make that happen :-)
And maybe it will be another feature for the deauther v3 :-D
I'll be describing my progresses in this Issue if can make some.
(Sorry for my terrible English)

WaGi-Coding, 2022-12-08:
> Yeah, argument values must be URL escaped. Thats why we have to use %20 instead of space. I think I can make some progress on sending the cli output as response. I am currently reading the [...]

Would be great if you manage something like that ^^ I am lacking the skills for this language and do not really know how i could even achieve this properly. Personally so far for just the deauther, i enjoy v2 more than v3, maybe i missed something with v3, but i was not able to set it up the way i had with v2. Anyway, would be great for any version. And no worries, my english isn't the best either as you might can see ._.

Greetings

ra1n08, 2022-12-10:
oh, im finding such thing, if this possible, it will helpful for some apps on android and ios device 
like click deauth button on app GUI (not web gui) and it send request https://<deauther_ip>/run?=deauth <wifi_name> <timeout>

ra1n08, 2022-12-10:
but i think need a esp8266 have >4mb flash for web_cli, web_gui and some json config file
that big

NurTasin, 2022-12-10:
Hey buddies, I have made some progresses towards making the web console thing real.
I have added a `/console` endpoint to the server, that sends the raw console output.
Here I ran the code through the `/run` endpoint
 
[image]

And after running that. I made a HTTP_GET request at `/console` endpoint. And the output looks like this

[image]

Now I just have to make a web page for the WebConsole using some HTML,CSS,JS. This might take some time.
But you guys can take a look at my [fork](https://github.com/NurTasin/esp8266_deauther) of this repo and try it your self if you want.

NurTasin, 2022-12-12:
# Added `/webconsole.html` endpoint.
Hey, I have added the necessary HTML,JS and CSS files. And now you can control the deauther through a web console.
For now it has some flaws. Like:
- It updates the console once in every 5 seconds. So it's not actually realtime.
- It only sends the command from the console if the Enter key is pressed. This works perfectly with PCs but doesn't work with Mobile Phones.

I'll fix these problems in next release. You can get the releases [here](https://github.com/NurTasin/esp8266_deauther/releases/tag/v2.7).
Make sure that you have starred [my fork](https://github.com/NurTasin/esp8266_deauther) of this project to stay updated.

ra1n08, 2023-01-15:
nice you're maked api for deauther, so now i can access to deauther via python code