SpacehuhnTech/esp8266_deauther #1629: Oled screen not working

MothMan13300, 2023-02-27:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

**Describe the error**

board is wired up correctly but the screen just displays a few pixels 

**Expected behavior**

screen supposed to display main menu for the deauther

**Screenshots**
[image]
[image]

**Environment (please complete the following information):**
 - OS: win 10
 - Browser chrome
 - Version v1
 - Hardware Node Mcu Esp8266

**Attempts**
changing the address changing in arduino and the screen type (ssd and ssh) and none seems to work

**Additional context**
Provide as much information as possible, better too much than too little!
If you don't use this template, your issue might be closed and tagged invalid!

micockrr, 2023-03-03:
similar issue, any fixes?

TriDEntApollO, 2023-04-03:
did you find any fix to this?

billypolidorou, 2023-09-06:
the fix is go to wifi on your phone and check for (pwned) connect on this wifi and password is (deauther) later go to google and search 192.168.4.1 press on accept and go to settings scroll down and you see desplay check it press save and reboot !!!

closed 2025-05-06 by stale[bot]