SpacehuhnTech/esp8266_deauther #1650: esp8266 serial monitor after flashing

testsesdfs, 2023-05-23:
After flashing the esp8266 with deauther this led on the board blinks and this spams on serial monitor.
```
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

Mounting SPIFFS...OK
Loading settings...OK
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP

[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: false
STARTED! \o/
2.6.1
```
[image]

Im using this board:
[image]

Serial monitor logs: https://pastebin.com/RHbdVUdd

testsesdfs, 2023-05-23:
uploading logs: https://pastebin.com/N1ZVBKPy

testsesdfs, 2023-05-23:
Okay I fixed it.

KEKKO96, 2023-08-06:
> 

how?

closed 2025-05-06 by stale[bot]