SpacehuhnTech/esp8266_deauther #1652: Deauther 2.6.2 on ESP8266 operating Buttons not working ?

yashgunjal04, 2023-06-14:
the config is default:
// ===== BUTTONS ===== //
#define BUTTON_UP 14 // D5
#define BUTTON_DOWN 12 // D6
#define BUTTON_A 2 // D4
#define BUTTON_B 0 // D3
but connecting to the GND is without any change.
The terminal command : screen mode buttontest
shows only false on all 4 buttons.
there is no difference between D5,D6 or D3 connected to the GND or not

shakri0, 2023-07-22:
K

kinite-gp, 2024-04-05:
Hello dear friends
I had the same problem, I made some changes in the files and now it seems to be working

You need to open the `A_config.h` file

In the first step, I commented this line `#define DEFAULT_ESP8266` and instead I got this from the comment `#define WEMOS_D1_MINI` because I use "wemos mini d1"

I commented all the lines below

```c
#ifndef BUTTON_UP
  #define BUTTON_UP 255
#endif // ifndef BUTTON_UP

#ifndef BUTTON_DOWN
  #define BUTTON_DOWN 255
#endif // ifndef BUTTON_DOWN

#ifndef BUTTON_A
  #define BUTTON_A 255
#endif // ifndef BUTTON_A

#ifndef BUTTON_B
  #define BUTTON_B 255
#endif // ifndef BUTTON_B
```

I added the following lines

```c
#define USE_DISPLAY true
#define DISPLAY_TIMEOUT 600
#define FLIP_DIPLAY false

#define SSD1306_I2C

#define I2C_ADDR 0x3C
#define I2C_SDA 5
#define I2C_SCL 4

#define BUTTON_UP 14
#define BUTTON_DOWN 12
#define BUTTON_A 13
#define BUTTON_B 15
```

Because my Oled was i2c, I used `#define SSD1306_I2C` line, otherwise you can remove other lines from the comment that are next to the same line.

Just make sure to enter the pin numbers correctly, change the values from `255` to the desired base

Pay attention that the bases for spi are different, just look carefully at the comments in the source code and you will understand

I'm sorry for the many explanations and my bad English, my language is not English
I hope I can help people who have the same problem