SpacehuhnTech/esp8266_deauther #1663: vendor_list_generator.py not working, file was missing from the url

ktauchathuranga, 2023-08-14:
**Describe the bug**
can't update the vendor list

**To Reproduce**
just run the [vendor_list_generator.py](https://github.com/SpacehuhnTech/esp8266_deauther/blob/v3/vendor_list_generator/vendor_list_generator.py)

Alliegaytor, 2023-08-15:
The URL in the script is dead, you need to change it to something alive.

I'm using [this archive.org URL](https://web.archive.org/web/20230328024857if_/https://gitlab.com/wireshark/wireshark/-/raw/master/manuf) for my personal fork.

Relevant pull request: https://github.com/SpacehuhnTech/esp8266_deauther/pull/1664

ktauchathuranga, 2023-08-15:
> The URL in the script is dead, you need to change it to something alive.
> 
> I'm using [this archive.org URL](https://web.archive.org/web/20230328024857if_/https://gitlab.com/wireshark/wireshark/-/raw/master/manuf) for my personal fork.
> 
> Relevant pull request: #1664

thank you!

Alliegaytor, 2023-08-15:
> > The URL in the script is dead, you need to change it to something alive.
> > I'm using [this archive.org URL](https://web.archive.org/web/20230328024857if_/https://gitlab.com/wireshark/wireshark/-/raw/master/manuf) for my personal fork.
> > Relevant pull request: #1664
> 
> thank you!

Maybe this issue should be kept open as it hasn't been fixed yet? I couldn't find any documentation that tells users to find their own vendor list, so the script should have a live URL as to not have people continuing to make issues about it.

Also, the list I provided won't get updates (it's an archive). So another one would have to be found eventually :)

closed 2024-03-30 by ktauchathuranga