SpacehuhnTech/esp8266_deauther #1671: 7 Pin SPI-Display Not Working

AxelGeneta00, 2023-09-24:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes/No

**Describe the error**
A clear and concise description of what the problem is.
What do you think causes it?

```
Error/Compile/Output Log
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: Win 10 pro
 - Browser Google
 - Version 2.6.0
 - Hardware Esp8266(NodeMcu)
**Attempts**
When using without display it's fine. But when i enabling the display its not opening, I already do the instruction but the oled wont work, if i try to set the config to SPI Display Example my nodemcu keeps restarting. I already try the oled display to my arduino so its working fine.

**Additional context**
Provide as much information as possible, better too much than too little!
If you don't use this template, your issue might be closed and tagged invalid!

marchewaaa333, 2023-10-03:
try changing yours A_config to [this] (https://github.com/marchewaaa333/deauther-fixSPI/blob/main/A_config.h) , it's a [wh1ter0z](https://github.com/wh1ter0z)
 version of config changed to SPI display by me

AxelGeneta00, 2023-10-17:
thanks bro its working but still have slight prob, the screen only display white but theres letters in the side, the screen is fine im using this with other module and diy project. maybe my wirng is wrong, can you help me with the wiring thanks

closed 2025-05-06 by stale[bot]