SpacehuhnTech/esp8266_deauther #1676: I2C SSD1106 Display is not working (the oled is working)

Doridori9992, 2023-10-29:
Hello, I bought I2C SSD1106 which is working with other codes but when I upload that code with configuring A_config the display behaves dumb and It refreshes every 10 seconds.

here are my circuits:
[https://ibb.co/WVrjHGT
https://ibb.co/yBSwZZL
https://ibb.co/NxKY009](url)

here is my config file:
`/* This software is licensed under the MIT License: https://github.com/spacehuhntech/esp8266_deauther */

#pragma once

#define ENABLE_DEBUG
#define DEBUG_PORT Serial
#define DEBUG_BAUD 115200

#define DEFAULT_ESP8266

// #define NODEMCU
// #define WEMOS_D1_MINI
// #define HACKHELD_VEGA
// #define DISPLAY_EXAMPLE_I2C
// #define DISPLAY_EXAMPLE_SPI

// #define MALTRONICS
// #define DSTIKE_DEAUTHER_V1
// #define DSTIKE_DEAUTHER_V2
// #define DSTIKE_DEAUTHER_V3
// #define DSTIKE_DEAUTHER_V3_5
// #define DSTIKE_D_DUINO_B_V5_LED_RING
// #define DSTIKE_DEAUTHER_BOY
// #define DSTIKE_NODEMCU_07
// #define DSTIKE_NODEMCU_07_V2
// #define DSTIKE_DEAUTHER_OLED
// #define DSTIKE_DEAUTHER_OLED_V1_5_S
// #define DSTIKE_DEAUTHER_OLED_V1_5
// #define DSTIKE_DEAUTHER_OLED_V2
// #define DSTIKE_DEAUTHER_OLED_V2_5
// #define DSTIKE_DEAUTHER_OLED_V3
// #define DSTIKE_DEAUTHER_OLED_V3_5
// #define DSTIKE_DEAUTHER_OLED_V4
// #define DSTIKE_DEAUTHER_OLED_V5
// #define DSTIKE_DEAUTHER_OLED_V6
// #define DSTIKE_DEAUTHER_MOSTER
// #define DSTIKE_DEAUTHER_MOSTER_V2
// #define DSTIKE_DEAUTHER_MOSTER_V3
// #define DSTIKE_DEAUTHER_MOSTER_V4
// #define DSTIKE_DEAUTHER_MOSTER_V5
// #define DSTIKE_USB_DEAUTHER
// #define DSTIKE_USB_DEAUTHER_V2
// #define DSTIKE_DEAUTHER_WATCH
// #define DSTIKE_DEAUTHER_WATCH_V2
// #define DSTIKE_DEAUTHER_MINI
// #define DSTIKE_DEAUTHER_MINI_EVO

// #define LYASI_7W_E27_LAMP
// #define AVATAR_5W_E14_LAMP

// Forces formatting of SPIFFS and EEPROM ot startup
// #define FORMAT_SPIFFS
// #define FORMAT_EEPROM

// Forces a reset of all settings at startup
// #define RESET_SETTINGS

// ========== CONFIGS ========== //

// https://github.com/spacehuhntech/hackheld
#if defined(HACKHELD_VEGA)
// ===== LED ===== //
[... 641 lines cut ...]
 #define I2C_SCL 4

 #define SPI_RES 5
 #define SPI_DC 4
 #define SPI_CS 15

   // ===== BUTTONS ===== //
 #define BUTTON_UP 255
 #define BUTTON_DOWN 255
 #define BUTTON_A 255
 #define BUTTON_B 255

   // ===== Reset ====== //
 #define RESET_BUTTON 5

   // ===== Web ===== //
 #define WEB_IP_ADDR (192, 168, 4, 1)
 #define WEB_URL "deauth.me"

 */

// ========== ERROR CHECKS ========== //
#if LED_MODE_BRIGHTNESS == 0
#error LED_MODE_BRIGHTNESS must not be zero!
#endif /* if LED_MODE_BRIGHTNESS == 0 */`

wryhode, 2023-10-30:
The way the display glitches, it may be due to interference in the I2C wires. Try moving wires that aren't for the display away from them. You could also try to separate the data (SDA) and clock (SCL) wires even more.

AxelGeneta00, 2023-11-14:
same prob, the 0.96 works fine but the 1.3 have glitches, the oled is also fine

AxelGeneta, 2023-11-15:
[image]
Replace this SD1306 bro with SH1106 I2c

closed 2025-05-06 by stale[bot]