SpacehuhnTech/esp8266_deauther #1679: Issue with connecting i2c screen to wemos d1 mini

soapy75, 2023-11-13:
I have succefully managed to activate the web based version, but after correctly connecting the wemos d1 mini to sd1306 oled, There is nothing on the display and the device seems to just be restarting as shown by the repeating startup messages in the serial monitor.

I have tried:

- switching 3v3 to 5v
- commenting out other elif statements that arent related to the example
- swapping out all components
- also  trying to also define it as wemos as well as i2c example

I think it may be to do with the fact that i need the correct i2c adress, but after trying a code online it didnt work and outputted no i2c device found

AxelGeneta, 2023-11-15:
Did you activate the screen thru settings in web?

soapy75, 2023-11-15:
> Did you activate the screen thru settings in web?

There doesnt seem to be a pwn network since im running the example i2c configuration if thats what you mean.

closed 2025-05-06 by stale[bot]