SpacehuhnTech/esp8266_deauther #1688: NodeMCU 1.0 bootlooping problem

SuchyChomik, 2024-01-19:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes, but there wasn't anyone with the same problem.

**Describe the error**
The ESP8266 resets every ~8 seconds. I don't know whats causing the problem. Down is the console output (from https://serial.huhn.me/)

```
 ets Jan  8 2013,rst cause:4, boot mode:(3,4)

wdt reset

load 0x4010f000, len 3584, room 16 

tail 0

chksum 0xb0

csum 0xb0

v2843a5ac

~ld

Mounting SPIFFS...OK

Loading settings...OK

Device names loaded from /names.json

SSIDs loaded from /ssids.json

Scan results saved in /scan.json

Serial interface enabled

Started AP

[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: false

STARTED! \o/

2.6.1
```

**To Reproduce**
Steps to reproduce the behavior:
1. Plug it into your computer
2. Go into the console
3. See error

**Expected behavior**
Expected it to not bootloop

**Environment (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - Hardware: DIY

**Attempts**
Reflashing the software a couple of times, searching the error.

**Additional context**
Used the bin file and the https://esp.huhn.me/ website for flashing the esp

moeburn, 2024-06-03:
The "Display Example I2C" board config is broken. Do not use it. Instead, under the "deauther config" menu, select the default "NodeMCU" at the top, and then in A_config.h, manually copy everything from the "Display Example I2C" section to the top, outside of its "#if defined" group.

ellensp, 2025-02-16:
Just struck this issue.

disabling #define DISPLAY_EXAMPLE_I2C
and adding the following directly under #define NODEMCU

```
#define SSD1306_I2C
#define I2C_ADDR 0x3C
#define I2C_SDA 5
#define I2C_SCL 4

``` 
Works fine.