SpacehuhnTech/esp8266_deauther #1689: Is there any way to use the deauther without a screen?

devicemodder, 2024-01-29:
> Please search for existing (open and closed) issues first to avoid duplicates.  
Also have a look at the [Wiki](https://github.com/spacehuhntech/esp8266_deauther/wiki).  

I have one of those wifi controlled plugs that use an ESP8266. I am installing the deauther on the plug and it does not have a screen. is there any way to control the deauther without the screen or buttons?

Neutrale35, 2024-01-29:
Just connect to pwned network and browse 192.168.4.1. You don't need a display. You can use your phone or computer

devicemodder, 2024-01-29:
Ah, very good. Now for the next bit... what part of the code can I modify
to add the relay function of the plug that I installed this on?

Ie, I would like to add a toggle to the web page/serial and ve able to use
the pushbutton on the plug. How easy would that be to do with the deauth
code?

closed 2025-05-08 by stale[bot]