SpacehuhnTech/esp8266_deauther #1690: Raspberry Pi Pico W support?

SerialDesignatorN, 2024-01-29:
**Do similar feature requests issues already exist (open and closed)?**
No.

**Is your feature request related to a problem? Please describe.**
I had a Pi Pico W (yes, it's with the antenna and WiFi) and I wonder if I can use it for deauthing to test its capabilities.

**Describe the solution you'd like**
To have a very stable support for the Pi Pico W

**Describe alternatives you've considered**
minigotchi - (FAILED)

**Additional context**

--

wryhode, 2024-02-19:
You could probably do something similar with a RPI-Pico W, but this project hasn't gotten any updates in a long while. Both because of that, and the fact that the ESP8266 handles WiFi differently, I think there might never be a RPI-Pico W build of this exact project.
I've tried to research a bit and it doesn't look like anyone has made a RPI-pico W deauther yet. It looks like you'll have to make this yourself, so maybe this article can help you get started? https://hackaday.com/2022/12/13/roll-your-own-wifi-driver-for-the-pico-w/

Wannacry1234455900, 2024-02-19:
Thank you beautiful 😻