SpacehuhnTech/esp8266_deauther #1696: Wpa3

smoochiee, 2024-03-01:
Unable to deauth wpa3 encryptions

wryhode, 2024-03-01:
Afaik WPA3 requires protected management frames to be used, which quite securely protects against deauth-attacks.
See "Protected Management Frames" in this article: https://blog.spacehuhn.com/deauth-attack-not-working

Also worth mentioning is that the ESP8266 can only use 2.4GHz networks. The access point might be 5GHz.

smoochiee, 2024-03-01:
> Afaik WPA3 requires protected management frames to be used, which quite securely protects against deauth-attacks.
> See: https://blog.spacehuhn.com/deauth-attack-not-working#heading-protected-management-frames
> 
> Also worth mentioning is that the ESP8266 can only use 2.4GHz networks. The access point might be 5GHz.

Okay ive read that 2.4ghz with wpa3 can be deauth...

closed 2024-03-01 by smoochiee