SpacehuhnTech/esp8266_deauther #1706: keep rebooting in loop

asg02fever, 2024-05-01:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

**Describe the error**
code uplaod successfully , but the serial monitor keep rebooting , cant see the ssid in my wifi page, i am on a node mcu esp12e i set the scree scl sda pins , my buttun pins , with bin file work great but screen is not set correctly so cant use it with bin 
```
Error/Compile/Output Log
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]
[image]

**Environment (please complete the following information):**
 - OS: [win 11 with cs drivers]
 - Browser [chrome but using ide]
 - Version [2.3.2]
 - Hardware [esp8266 node mcu esp12e module screen inboard sda gpio14 d6 , scl gpio12 d5 button on dpio 5/4/0 pin d1/2/3 ]

**Attempts**
almost everything of my small knowledfe

**Additional context**
Provide as much information as possible, better too much than too little!
If you don't use this template, your issue might be closed and tagged invalid!

moeburn, 2024-06-03:
I have the fix.  The "Display Example I2C" board config is broken.  Do not use it.  Instead, under the "deauther config" menu, select the default "NodeMCU" at the top, and then in A_config.h, manually copy everything from the "Display Example I2C" section to the top, outside of its "#if defined" group.

Mehrdadmji, 2024-06-25:
> I have the fix. The "Display Example I2C" board config is broken. Do not use it. Instead, under the "deauther config" menu, select the default "NodeMCU" at the top, and then in A_config.h, manually [...]

Hello, my code has a similar problem, my oled screen does not show anything, can you help me or share the fixed code with me bro?

asg02fever, 2024-06-26:
hey man can u add me on discord keebasg i will send you the fixed code

closed 2024-06-29 by asg02fever

mannapijush1coretech, 2024-07-10:
same problem even i place display example #define Display Example I2C outside of that if region problem is same!
even i flashed bin files in the section releases ! display example i2c problem stays same ! 
if u want to share code then it will be helpful

```
Mounting SPIFFS...OK
Loading settings...OK
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: false
STARTED! \o/
2.6.1

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
```

asg02fever, 2024-07-10:
add me on discord i will give you the code