SpacehuhnTech/esp8266_deauther #1708: Can i Add an antenna or an Mod for 5Ghz Support? (esp8266)

eminsk45, 2024-06-12:
Hello, i have made my own Deauther Lately and added Battery on it and so and i saw on youtube a Video of a Guy adding an Antenna on the Esp8266 Deauther. Someone asked in the comments if it was possible to Deauth 5Ghz Network and he Replied that it is possible with some Mods. And i wanted to aks what mods are nedeed for the esp8266 so that it can Deauth 5ghz Wifi?
Thank You.

Video Link: https://www.youtube.com/watch?v=QDv3r4PMKIw

vaibhav88614, 2024-06-28:
No.You cannot  Deauth a 5Ghz radio frequency band using 2.4Ghz radio frequency band (esp8266).I would suggest you to upgrade to any other similar device which supports 5Ghz(ex:-esp32) and upload the required code to get things done.If you need any help with code or getting started just ping me.

Ki4o112, 2024-08-10:
Dose ESP32 support 5Ghz ? I was going to buy this board but i don't know what is it : https://prnt.sc/WVxpkDvDpZA7

closed 2025-05-08 by stale[bot]