SpacehuhnTech/esp8266_deauther #1734: 0.91 inch isn't working

Harshu1000-7, 2024-09-05:
> Have you searched for existing (open and closed) issues describing the same problem?

No

**Describe the error**
0.91 inch display won't just start working

```
Error/Compile/Output Log
Nothing 
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Expected to see display output and showing attacks "death probe and beacon"

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

 - Hardware [e.g. DSTIKE, DIY]
 - esp8266

**Attempts**
Changing pin numbers

**Additional context**
I've seen videos of people directly plugging 0.91 inch display after unplugging 0.96 inch ssd1306 display the 0.91 inch as mentioned on data sheet has the same I ssd1306 but still the display won't work

Sudo-CHOPP, 2025-02-14:
Any luck? I cant get mine working either. Im brand spanking new to this kind of stuff though so I have no idea what I'm doing. Just kinda tinkering.

Harshu1000-7, 2025-02-14:
> Any luck? I cant get mine working either. Im brand spanking new to this kind of stuff though so I have no idea what I'm doing. Just kinda tinkering.

yes i got the fix, you have to connect the display the way it is, and the display can be enabled by going on webserver settings and enabling it, its disabled by default