SpacehuhnTech/esp8266_deauther #1743: Stuck at Initializing - Deauth app

elk1007, 2024-10-03:
**> Have you searched for existing (open and closed) issues describing the same problem?**
Yes

**Describe the error**
When I open the Deauth app on Flipper, it gets stuck at Initializing... 

I was able to successfully flash Deauth onto my D1 mini V4 (using flash tool QOUT method).
The module is detected, and I can even flash using ESP Flasher. 
Selecting Enter Bootloader in ESP Flasher creates the SSID pwned and I am able to connect. 

I must be overlooking something simple, as the deauth app can't go past Initializing.... 

 

```
**Error/Compile/Output Log**

Please let me know where I can find a relevant log file to this issue. I will post it. 
```

To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Expected behavior
Screen should go past "initializing..."

Environment (please complete the following information):
-Flipper Zero
-Unleashed firmware w/preinstall and 'flipper-z-f7-full-local' firmware which shows deauth in main menu (tried both, same issue "initializing")
-D1 Mini Ver4 

Attempts
I have tried to swap TX/RX pins. 
I have tried to connect/disconnect Pin 0 to ground. This was required to flash the chip. 
I have tried reinstalling firmware and resetting flipper. 

Any input, suggestion, or guess, is appreciated.

elk1007, 2024-10-04:
https://github.com/SequoiaSan/FlipperZero-Wifi-ESP8266-Deauther-Module/issues/10#issuecomment-2386093446

Resolved

closed 2024-10-04 by elk1007

TickTastley, 2025-02-04:
How did you resolve it? I spent my pocket money for this, I need to know immediately, my dad will kill me :D