SpacehuhnTech/esp8266_deauther #1747: Is there a way to customize web interface?

TunaFish111, 2024-10-24:
Hello im kinda new to this stuff. But is there a way to change the web colors? Its totally fine if i cant but it would be really cool if i could make the whole web interface red or smtn. Thanks!

IDknoobcoder, 2024-10-24:
Yeah you could go to web interface folder if u know how to code in html and css also java u could easily do that

TunaFish111, 2024-10-24:
> Yeah you could go to web interface folder if u know how to code in html and css also javascript u could easily do that

I changed some stuff in the html and css code. I put it in the .gz files but when i run it its always the default web interface please help.

IDknoobcoder, 2024-10-24:
Pls tell path of gz file

TunaFish111, 2024-10-24:
> Pls tell path of gz file

It was in deauthner\data\web pretty sure im not home rn

IDknoobcoder, 2024-10-24:
Wht r u doing like changing the code then compressing it to gz file I think we need to code the web interface files then change or compress all those files and save in gz folder

TunaFish111, 2024-10-24:
> Wht r u doing like changing the code then compressing it to gz file I think we need to code the web interface files then change or compress all those files and save in gz folder

I changed the web interface files. What do i do now?

IDknoobcoder, 2024-10-24:
Maybe use 7zip or try winrar

IDknoobcoder, 2024-10-24:
Also could u help me in my issue #1746

TunaFish111, 2024-10-24:
> Maybe use 7zip or try winrar

What i did was customize the html and css code, save it and then compress it using 7zip and then change the files and upload it in the esp. I did this prior to these messages and it didnt seem to work. Please tell me what i did wrong.

IDknoobcoder, 2024-10-24:
Some error messages if u could tell

TunaFish111, 2024-10-24:
> Some error messages if u could tell

Will you be online in around 30 mins? Im heading home rn. Ill try again. However im pretty sure there wasnt any error messages. I uploaded the code everything went fine but the interface wouldnt change

IDknoobcoder, 2024-10-24:
Ok

TunaFish111, 2024-10-24:
> Ok

Any chance you could try and tell me how you did it? Im sorry for causing you trouble

IDknoobcoder, 2024-10-24:
Nah I can’t right now

TunaFish111, 2024-10-24:
> Nah I can’t right now

Thats fine. So i should just compress the html files, and then change them with the ones in the web file?

IDknoobcoder, 2024-10-24:
Yeah change the files that u made (custom version compressed) in all the gz folder hope it would help u

TunaFish111, 2024-10-24:
> Yeah change the files that u made (custom version compressed) in all the gz folder hope it would help u

Thanks dude and sorry for wasting your time

IDknoobcoder, 2024-10-24:
Nah bro don’t sorry I had time that is why replied 👊

IDknoobcoder, 2024-10-26:
Did that work

IDknoobcoder, 2024-10-31:
Does it work

EurekaEffect, 2025-01-17:
> Does it work

No it doesn't, I have the same issue