SpacehuhnTech/esp8266_deauther #1748: Upload error with NodeMCU

thijbel, 2024-11-01:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

**Describe the error**
whenever i try to upload the deauther code to my nodemcu board it just throws the same error that i show on the image

```
Sketch uses 829464 bytes (79%) of program storage space. Maximum is 1044464 bytes.
Global variables use 32132 bytes (39%) of dynamic memory, leaving 49788 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port COM3
Traceback (most recent call last):
  File "C:\Users\vlxdh\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.7.5/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/vlxdh/AppData/Local/Arduino15/packages/deauther/hardware/esp8266/2.7.5/tools/esptool\esptool.py", line 2889, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:/Users/vlxdh/AppData/Local/Arduino15/packages/deauther/hardware/esp8266/2.7.5/tools/esptool\esptool.py", line 237, in __init__
    self._port = serial.serial_for_url(port)
  File "C:/Users/vlxdh/AppData/Local/Arduino15/packages/deauther/hardware/esp8266/2.7.5/tools/pyserial\serial\__init__.py", line 88, in serial_for_url
    instance.open()
  File "C:/Users/vlxdh/AppData/Local/Arduino15/packages/deauther/hardware/esp8266/2.7.5/tools/pyserial\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM3': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
```

**Expected behavior**
Expected to flash correctly the board ...

**Screenshots**
[image]

**Environment (please complete the following information):**
 - OS: Window 11
 - Browser Chrome
 - Version lastet maybe ...
 - Hardware DIY

**Attempts**
I tried every way but nothing seems to work

**Additional context**
Help guysssssssss

IDknoobcoder, 2024-11-01:
Reinstall the ch340 driver while esp plugged in

thijbel, 2024-11-02:
> Reinstall the ch340 driver while esp plugged in

it doesn't work

closed 2025-06-27 by stale[bot]