SpacehuhnTech/esp8266_deauther #1750: Buttons not working with Deauther

BigPszemek, 2024-11-02:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

**Describe the error**
The Buttons do not work with the deauther code but they do work with my code. I am in no way educated on the topic, and i dont really know what can be causing it. I assume its because i have connected it in a different way than what the code was written for.
I have tested the buttons with this code:

const int BUTTON_UP = 14; 
const int BUTTON_DOWN = 12;
const int BUTTON_A = 13;

void setup() {
    Serial.begin(115200);
    pinMode(BUTTON_UP, INPUT_PULLUP);
    pinMode(BUTTON_DOWN, INPUT_PULLUP);
    pinMode(BUTTON_A, INPUT_PULLUP);
}

void loop() {
    if (digitalRead(BUTTON_UP) == LOW) {
        Serial.println("Button UP pressed");
    }
    if (digitalRead(BUTTON_DOWN) == LOW) {
        Serial.println("Button DOWN pressed");
    }
    if (digitalRead(BUTTON_A) == LOW) {
        Serial.println("Button A pressed");
    }

    delay(200); 
}
Video of the code working:
[Screencast from 2024-11-02 19-01-35.webm](https://github.com/user-attachments/assets/2a5e3993-6f9d-476c-8389-60b2292b722c)

**Expected behavior**
Buttons should be working i guess

**Screenshots**
thats how it looks like(might be hard to read):
[image]
and thats how the buttons are set up:
[image]

**Environment (please complete the following information):**
 - OS: [Both Windows 11 and Debian 12]
 - Version [Deauther V2]
 - Hardware [Esp8266 i guess even though it says NODEMCU on the bottom]

**Attempts**
i've tried many cable combinations and altering the deauther code but as im new both did not work.

**Additional context**
so when theres no resistor the pin reads the button as a different state but when resistor added the state is not fluid anymore, works with my code but not deauther.

IDknoobcoder, 2024-11-05:
Go to A_ config just comment nodemcu and uncomment  any dstike version that matches your buttons and screen configuration I recommend to uncomment the distike mini at line 45

BigPszemek, 2024-11-06:
> Go to A_ config just comment nodemcu and uncomment any dstike version that matches your buttons and screen configuration I recommend to uncomment the distike mini at line 45

BigPszemek, 2024-11-06:
> Go to A_ config just comment nodemcu and uncomment any dstike version that matches your buttons and screen configuration I recommend to uncomment the distike mini at line 45

accidentally closed the thread, |

i feel like giving up on this and just starting to do something on my own.
tried changing it and it didnt work (:

IDknoobcoder, 2024-11-07:
Bro don’t give up https://github.com/SpacehuhnTech/esp8266_deauther/releases download the bin file an flash it with esp web tool use deauther mini or try other if it doesn’t work

IDknoobcoder, 2024-11-07:
#1573 u can refer this conversation also

BigPszemek, 2024-11-07:
> #1573 u can refer this conversation also

hey, so i got a different board and used your previous advice (45 line) and it works now. thanks alot man hope you do well :)

IDknoobcoder, 2024-11-07:
Ur welcome

iprosenjit9, 2025-05-10:
On line 574  to 584 change 255 to respective button pin value
-------From this-----------
#ifndef BUTTON_UP
  #define BUTTON_UP 255
#endif // ifndef BUTTON_UP

#ifndef BUTTON_DOWN
  #define BUTTON_DOWN 255
#endif // ifndef BUTTON_DOWN

#ifndef BUTTON_A
  #define BUTTON_A 255
#endif // ifndef BUTTON_A

--------To this------

#ifndef BUTTON_UP
  #define BUTTON_UP 14
#endif // ifndef BUTTON_UP

#ifndef BUTTON_DOWN
  #define BUTTON_DOWN 12
#endif // ifndef BUTTON_DOWN

#ifndef BUTTON_A
  #define BUTTON_A 13
#endif // ifndef BUTTON_A