SpacehuhnTech/esp8266_deauther #1757: Problem with the password

WavierPigeon261, 2024-12-15:
Hello,
When I installed this deauther code in my nodemcu esp8266 like 3 months ago, I had changed its said and password of that access point. Now I don't remember it. Can you help me in retrieving the password of that access point? Or else is there any option to reset the code completely? Please help me.

Arnab417, 2025-01-22:
re install the firmware

EinsTech-afk, 2025-03-07:
Try To Reflash The Program In The Esp8266 it will Reset everything hope it helps😉

WavierPigeon261, 2025-03-08:
> Try To Reflash The Program In The Esp8266 it will Reset everything hope it helps😉

I had tried this too, but it didn't work.
Thanks btw 😊

WavierPigeon261, 2025-03-08:
> re install the firmware

I did it, but I didn't work
Thanks 👍

EinsTech-afk, 2025-03-09:
Oof Try Deleting the File Of the Deauther and Download a New and Fresh On from The page 2.6.1 and Recommend Watch The tutorial On installation😁

WavierPigeon261, 2025-03-21:
Hello everyone,
Thank you for your help, **I tried resetting my esp8266 and now I can use the default credentials.** Once again thank you.

closed 2025-03-21 by WavierPigeon261