SpacehuhnTech/esp8266_deauther #1765: PWNED network

Lunar-Chaos, 2025-02-16:
> Have you searched for existing (open and closed) issues describing the same problem?
Yes

**Describe the error**

Hello, im using a weemos D1 mini V4 board, the issue is that i cannot connect to his wifi, on the serial i can see that he finds some access points, but i cannot open the website anyway, any help?
Error/Compile/Output Log
```
LOGS: 

11:51:52.957 -> # sysinfo
11:51:52.957 -> [======== SYSTEM INFO ========]
11:51:52.957 -> RAM usage: 51008 bytes used [63%], 30912 bytes free [37%], 81920 bytes in total
11:51:52.957 -> 
11:51:52.957 -> Current WiFi channel: 1
11:51:52.957 -> AP MAC address: ca:c9:a3:3a:d9:d8
11:51:52.957 -> Station MAC address: 00:26:73:1e:ef:43
11:51:52.957 -> SPIFFS: 24576 bytes used [37%], 40960 bytes free [62%], 65536 bytes in total
11:51:53.005 ->         block size 4096 bytes, page size 256 bytes
11:51:53.005 -> Files: 
11:51:53.005 ->   autostart.txt 19 bytes
11:51:53.005 ->   names.json 2 bytes
11:51:53.005 ->   scan.json 91 bytes
11:51:53.005 ->   ssids.json 318 bytes
11:51:53.005 -> [WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: false

**Environment (please complete the following information):**
 - OS: Windows 10
 - Browser chrome
 - Hardware Weemos D1 Mini V4

**Attempts**
using different versions, scketch and bin flash, reset sketch

Lunar-Chaos, 2025-02-16:
the error i get while connecting to wifi is: Failed connection

diegoespindola, 2025-04-08:
Exactly the Same issue over here, Using chinese wemos d1 mini pro
installed deauther v2 branch with android IDE in ubuntu